adafruit/Adafruit_CircuitPython_seesaw #2: Add CircuitPython HW API compatible classes

tannewt, 2017-12-13:
SeeSaw functionality should be expose by objects that mimic the native HW API. For example, digital pin reading and writing should match [DigitalInOut](https://circuitpython.readthedocs.io/en/2.x/shared-bindings/digitalio/DigitalInOut.html).

This should include:

- [DigitalInOut](https://circuitpython.readthedocs.io/en/2.x/shared-bindings/digitalio/DigitalInOut.html)
- [PWMOut](https://circuitpython.readthedocs.io/en/2.x/shared-bindings/pulseio/PWMOut.html)
- [AnalogIn](https://circuitpython.readthedocs.io/en/2.x/shared-bindings/analogio/AnalogIn.html)
- [NeoPixel](https://circuitpython.readthedocs.io/projects/neopixel/en/latest/api.html)
- [nvm](https://circuitpython.readthedocs.io/en/2.x/shared-bindings/nvm/ByteArray.html)

ladyada, 2018-05-01:
PWMout channel has been added :)

deanm1278, 2018-05-01:
DigitalInOut and AnalogIn have been added as well

deanm1278, 2018-05-02:
NeoPixel has been added as well

closed 2018-05-02 by deanm1278