adafruit/Adafruit_CircuitPython_seesaw #11: Button pressing crash w motors and servo

jedgarpark, 2018-06-12:
I'm crashing my bumper bot! Trying to use button a to start tt motors and turn servo once, then button b to stop motors and turn servo once. sometimes works, but often crashes the board when pressing the button b in particular. Seems to be worse if I rapidly press buttons.

 This will eventually also need to respond to accelerometer "shake" when it hits a wall, telling motors to back up, skid steer turn, then go forward again, repeat. 

Here it is in action from my livestream last week using MakeCode: https://youtu.be/12FeLD6vjQ8?t=1h7m36s  (Incidentally, the MakeCode version has stopped working as well and I have an issue in to that team on it. Weird!)

This is the error:
`Traceback (most recent call last):
  File "main.py", line 94, in <module>
  File "adafruit_motor/motor.py", line 79, in throttle
  File "adafruit_seesaw/pwmout.py", line 55, in duty_cycle
  File "adafruit_seesaw/seesaw.py", line 302, in analog_write
  File "adafruit_seesaw/seesaw.py", line 376, in write
  File "adafruit_seesaw/seesaw.py", line 376, in write
  File "adafruit_bus_device/i2c_device.py", line 102, in write
OSError: [Errno 5] Input/output error`

This is the code:
`# Bumper Bot for Crickit and CPX
# uses two TT motors for wheels plugged into Motor 1 and Motor 2,
# a micro servo for flag plugged into Servo 1
# NeoPixels for status
# Beep over the speaker

import time
import audioio
import board
from adafruit_seesaw.seesaw import Seesaw
from adafruit_seesaw.pwmout import PWMOut
from adafruit_motor import motor
from adafruit_motor import servo
import neopixel

from digitalio import DigitalInOut, Direction, Pull
from busio import I2C

button_a = DigitalInOut(board.BUTTON_A)
button_a.direction = Direction.INPUT
button_a.pull = Pull.DOWN

button_b = DigitalInOut(board.BUTTON_B)
button_b.direction = Direction.INPUT
button_b.pull = Pull.DOWN

# Create seesaw object
i2c = I2C(board.SCL, board.SDA)
seesaw = Seesaw(i2c)

print("Bumper Bot")

RED = 0x100000  # (0x10, 0, 0) also works
GREEN = (0, 0x10, 0)
BLACK = (0, 0, 0)
BLUE = (0, 0, 0x10)
YELLOW = (0x10, 0x10, 0)
PURPLE = (0x10, 0, 0x10)

pixels = neopixel.NeoPixel(board.NEOPIXEL, 10, brightness=0.2)
pixels.fill((YELLOW))  # fill yellow at startup
pixels.show()

# Create two motors on seesaw PWM pins 22 & 23, Crickit Motor 1 port
motor_a = motor.DCMotor(PWMOut(seesaw, 22), PWMOut(seesaw, 23))
motor_b = motor.DCMotor(PWMOut(seesaw, 19), PWMOut(seesaw, 18))

# Create servo object
pwm = PWMOut(seesaw, 17)     # Servo 1 is on s.s. pin 17
pwm.frequency = 50           # Servos like 50 Hz signals
my_servo = servo.Servo(pwm)  # Create my_servo with pwm signal
my_servo.angle = 90  # set the flag to middle position

############### User variables
run_time = 6
speed = 0.3

############### Music
cpx_audio = audioio.AudioOut(board.A0)  # speaker out on Crickit
def play_file(wavfile):
    with open(wavfile, "rb") as f:
        wav = audioio.WaveFile(f)
        cpx_audio.play(wav)
        while cpx_audio.playing:
            pass

# Wait before startup
time.sleep(1.5)
wav_file_name = "Boing.wav"
play_file(wav_file_name)

while True:

    while button_a.value:
        time.sleep(0.02)
        while button_a.value:  # wait for button release
            pass
        time.sleep(0.02)
        my_servo.angle = 180  # rotate the flag forward
        time.sleep(0.02)
        motor_a.throttle = (speed * -1)  # run both motors forward
        time.sleep(0.02)
        motor_b.throttle = speed
        time.sleep(0.02)
        # time.sleep(run_time)

    if button_b.value:
        time.sleep(0.02)
        while button_b.value:  # wait for button release
            pass
        time.sleep(0.02)
        motor_a.throttle = 0  # stop both motors
        time.sleep(0.02)
        motor_b.throttle = 0
        time.sleep(0.02)
        my_servo.angle = 90  # rotate the flag to center
        time.sleep(0.02)`

ladyada, 2018-06-12:
@jedgarpark thx, plz use ```'s for code blocks, or you can use the GUI <> to insert code:

[image]

tannewt, 2018-06-12:
It'd be interesting to determine if this is a SeeSaw firmware problem or a MakeCode/CircuitPython problem. Since it happens with both it makes me suspect its the SeeSaw firmware.

ladyada, 2018-06-12:
@jedgarpark for the code you put above, is that the broken one or the one you got working?

deanm1278, 2018-06-12:
hey @jedgarpark can you try doing this with no motors plugged in and make sure it still crashes?

deanm1278, 2018-06-12:
we've hashed this out to be a power supply issue and not code related.

closed 2018-06-12 by deanm1278

arnieforte, 2018-10-13:
I am currently having this issue with a feather cricket. What is the fix? It says power supply but is that the power that is getting attached to the board or the power supply circuitry on the board? One interesting thing is that I am catching the IO Error and if I reset the cricket board, the motors start working again. 

I also tried the test of disconnecting the motors from the board and it does not crash.

ladyada, 2018-10-13:
hiya - arnie please post in the forums since this is a tech quesion not an issue - but try fresh batteries too!