adafruit/Adafruit_CircuitPython_seesaw #30: Support for PulseIn

wallarug, 2019-05-04:
Hi,

Is there support for pulseio.PulseIn somewhere in SeeSaw?

I can see that it is mentioned but I am unsure how to use it with SeeSaw.  

There is CircuitPython support.  I have this example working in CircuitPython, however, not worked out how to port it across to SeeSaw.  If someone could point me in the right direction or say: "implement it please", that would be great!

```
from pulseio import PWMOut, PulseIn, PulseOut
from adafruit_motor import servo
from array import array
import time

steering_channel = PulseIn(board.RCH1)
throttle_channel = PulseIn(board.RCH2)
pwm1 = PWMOut(board.SERVO1, duty_cycle=2 ** 15, frequency=50)
pwm2 = PWMOut(board.SERVO2, duty_cycle=2 ** 15, frequency=50)
steering_servo = PulseOut(pwm1)
throttle_servo = PulseOut(pwm2)

while True:    
    while len(throttle_channel) == 0:
        pass
    throttle_channel.pause()  
    throttle_servo.send(array('H', [(throttle_channel[0])]))
    throttle_channel.resume()
    time.sleep(0.01)

```

Thanks in advance.

ladyada, 2019-05-04:
implement it please :)

wallarug, 2019-05-05:
No worries 😜

closed 2021-04-23 by ladyada