adafruit/Adafruit_CircuitPython_seesaw #32: Neopixel White value is ignored when RGB set to 0

EricGallimore, 2019-06-07:
If you want to make an RGBW pixel white by setting a white value like this:
neopixels.fill((0,0,0,255))

it will instead set the W value to 0.  The offending logic is here:
```
        # If all components are the same and we have a white pixel then use it
        # instead of the individual components.
        if self._bpp == 4 and r == g == b:
            w = r
            r = 0
            g = 0
            b = 0
```

Perhaps this should change, such that the W pixel value is only inferred from RGB values if the W value is 0?  The `if` statement could become
` if self._bpp ==4 and r==g==b and w==0:`

ladyada, 2019-06-10:
we dont use RGBW pixels so much, but if you're up for trying and fixing we'd love a PR!

zostay, 2019-07-09:
This is a really interesting problem. The suggested correction will fix the problem case in particular, but it will still cause problems when someone really wants to directly control all 32-bits of color. I think a better solution is:

1. Implement the change suggested by @EricGallimore 

2. Add an additional check that disables the code all together in case someone wants direct control over the LEDs without any interference.

So the code becomes something like:

```
if self._bpp == 4 and self._casual == 1 and r==g==b and w==0:
```

(`_casual` is maybe not the right name, but I had to pick something...)

Then, if someone flips casual control bit to 0, this logic turns off and they can control all RGBW pixels without interference by the `__setitem__` smart logic. That way if someone wants to burn the watts to turn their 8x8 RGBW pixel grid full on, they are welcome to do so. I hope they have a heat sink and a pair of solar glasses.

caternuson, 2019-08-21:
Is this automagic behavior more of hindrance than a convenience? Another idea might be to get rid of it and just require explicitly setting what you want.

ladyada, 2019-08-21:
thanx!

closed 2019-08-21 by ladyada

caternuson, 2019-08-21:
@EricGallimore Thanks for pointing this out. Please try updating to the 1.5.2 version of the library:
https://github.com/adafruit/Adafruit_CircuitPython_seesaw/releases/tag/1.5.2

PR #40 (2019-08-21): Fix white channel getting set 0.
Fixes #32. I went with @EricGallimore 's suggestion.

Tested with Crickit HAT on RPi:
```python
pi@crickit:~ $ python3
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> from adafruit_seesaw.seesaw import Seesaw
>>> from adafruit_seesaw.neopixel import NeoPixel
>>> pixels = NeoPixel(Seesaw(board.I2C()), 20, 8, bpp=4, pixel_order=(1, 0, 2, 3))
>>> pixels.fill(0)
>>> pixels.fill((0,0,0,255))
```