adafruit/Adafruit_CircuitPython_seesaw #33: Invalid soil temperature readings

jerryneedell, 2019-06-22:
I'm not sure where this issue belongs, but I'd appreciate it if someone and see if they can reproduce it.
This was originally reported here https://github.com/adafruit/circuitpython/issues/1912  butI think it better addressed independent from that issue. I'n not sure if it is a CP or a seesaw issue.

I had been running this sensor successfully up to a few weeks ago but now when I try to run the soil_simpletest.py I get a temperature reading of 16383 consistently. The moisture value looks OK. If I hold my finger on the sensor SAMD chip, there is sometime a small change in the temperature, but it is still meaningless. It should be ~26. where it is located.

Observation: 16383 would be returned if the value read was 0x3ffffff.

I see this on two different sensors.
I tried the sensors with Arduino and they return good temperature and moisture values.

Note, I am running CP 4.10 beta (current master on 6/20)

I get the same results on a feather_m4_express or Pyportal with the I2C interface set to 3.3 V

Note: this all started when I tried using the PyPortal with the I2C interface at the default 5V. When I saw the bad temperature readings, I thought it was due to the I2V voltage and switched to a PyPortal with the jumper set for 3.3V. The problem has remained.
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.1.0-beta.0-31-g3863a6ab8 on 2019-06-20; Adafruit PyPortal with samd51j20
>>> 
>>> 
>>> import soil_simpletest
temp: 16383.0  moisture: 338
temp: 16383.0  moisture: 351
temp: 16383.0  moisture: 351
temp: 16383.0  moisture: 351
```

ladyada, 2019-06-22:
@caternuson could be related to https://github.com/adafruit/Adafruit_CircuitPython_seesaw/issues/23 ?

jerryneedell, 2019-06-22:
I tried some simple changes and it looks like the I2C read for the temperature value is just flaky. I'll do some more testing and provide more details tomorrow.

It would be good to know if anyone can reproduce this issue....Just run soil_simpletest.py  
moisture is fine - temperature is bogus.

jerryneedell, 2019-06-23:
This is all it needed: longer delay in I2C read
```
diff --git a/adafruit_seesaw/seesaw.py b/adafruit_seesaw/seesaw.py
index afdaf19..6be55df 100644
--- a/adafruit_seesaw/seesaw.py
+++ b/adafruit_seesaw/seesaw.py
@@ -328,7 +328,7 @@ class Seesaw:
 
     def get_temp(self):
         buf = bytearray(4)
-        self.read(_STATUS_BASE, _STATUS_TEMP, buf)
+        self.read(_STATUS_BASE, _STATUS_TEMP, buf, .005)
         buf[0] = buf[0] & 0x3F
         ret = struct.unpack(">I", buf)[0]
         return 0.00001525878 * ret
```
PR coming after a few more tests

closed 2019-06-23 by ladyada

caternuson, 2019-06-26:
Ugh. Just now seeing this for some reason. Sorry.

Seems to make sense that a delay was needed. The `read_moisture` was already doing that with a value of `0.005`. So guess `read_temp` needed it also. Nice find @jerryneedell 

@ladyada But not sure if it relates to that spurious moisture reading issue, which persisted even with bumps in the delay. Could be related in some tangential way though - like deeper in seesaw?

ladyada, 2019-06-26:
possibly but i dont have cycles to debug :/ and this is a good enough fix...

PR #34 (2019-06-23): add delay to I2Cread for temperature reading
temperature reading had been working in the past, but now temperature readings were not being returned correctly. Not sure what changed but increasing the delay for the read fixed the problem.
This delay was already used for the moisture reading.

fixes #33 

Is this possibly due to recent "speedups" in CP in general?  Will other seesaw applications run into similar issues?  Is the default delay of .001 sufficient?

tested on feather_m4_express and pyportal