adafruit/Adafruit_CircuitPython_seesaw #35: What is pin_mode_bulk_b for?

zostay, 2019-07-08:
Okay, so I'm looking through the source code here because I wanted to understand how SeeSaw works. I was thinking of offering an improvement to `pin_mode_bulk`/`pin_mode_bulk_b` because I'm a software developer by day and that much duplicated code gives me the screamin' willies. However, before I can do that, I need to understand what is going on.

My confusion stems from what I read the SeeSaw data sheet (the only description of the SeeSaw I2C protocol I could find) versus what `pin_mode_bulk_b` appears to do. My understanding from the data sheet is that the GPIO interface is 32-bit. However, `pin_mode_bulk_b` is clearly performing a 64-bit operation by sending 8 bytes rather than 4.

My hypothesis, then, is that `pin_mode_bulk` will set the pin mode of "Bank A" of 32 pins, which is probably the only bank in most applications. Next, the `pin_mode_bulk_b`sets the pin mode for "Bank B" of 32 pins.  Over the wire the choice of bank is made based on whether the 4 bytes sent are all 0. If any bit is set, then Bank A is setup from the data given. If no bit is set, then another 4 bytes are expected and those are used to setup Bank B. (Corollary: If my hypothesis is true, one must be careful not to call `pin_mode_bulk` directly with all zeros in the bytearray or you may end up with an out-of-phase SeeSaw controller.)

In summary, you can work with up to 64 pins, but only in banks of 32 at a time. How close am I?

ladyada, 2019-07-08:
yeah, there's two PORTs on the samd chips we support, PA00-PA31 and PB00-PB31 - internally to the chip you can write all 32 bits as an atomic operation, and for reading multiple pins, its faster to do it in 'bulk' :)

zostay, 2019-07-09:
Cool beans.

closed 2019-07-09 by zostay