adafruit/Adafruit_CircuitPython_seesaw #37: Pin mappings are space efficient, but time inefficient

zostay, 2019-07-09:
When SeeSaw sends signals over the wire it uses a channel offset for some of the operations, which is calculated using the index of the pin in the pin mapping list. Here's the code for one such operation:

```
        self.read(_ADC_BASE, _ADC_CHANNEL_OFFSET + self.pin_mapping.analog_pins.index(pin), buf)
```

The way this works allows the pins to be stored with O(n) efficiency at the average cost in time to look up also being O(n) efficiency. This means that It takes several cycles longer to send messages for the pins listed later in the mapping than those at the start.

Also, as a weird side effect, if someone accidentally supplies a pin number that is not in the pin mapping, you could end up with undefined behavior because the index will be returned as -1.

I would suggest a better mechanism might to use a dictionary to map the pin numbers to their offsets. For example, the `analog_pins` for the Crickit would become:

```
    analog_pins = {_CRICKIT_SIGNAL1: 0, _CRICKIT_SIGNAL2: 1,
                   _CRICKIT_SIGNAL3: 2, _CRICKIT_SIGNAL4: 3,
                   _CRICKIT_SIGNAL5: 4, _CRICKIT_SIGNAL6: 5,
                   _CRICKIT_SIGNAL7: 6, _CRICKIT_SIGNAL8: 7)
```

And the lookup above would become:

```
        self.read(_ADC_BASE, _ADC_CHANNEL_OFFSET + self.pin_mapping.analog_pins.get(pin, -1), buf)
```

I have preserved the same weird behavior as before by using a -1 default just to maintain functional identity with the original operation, but it could be something more sensible or bounds checked instead.

The storage efficiency of this is now worse than O(n) because there's an internal hash table, but the lookups will now be O(1) for all pins, meaning all pins have the same efficiency. It will probably be slower for the early pins, but much faster for the later pins and on average better, even for only a handful of pins.

ladyada, 2019-07-09:
hiya - if n is 8 it doesnt matter what O() it is. we optimize for reducing memory footprint (which is in short supply)

zostay, 2019-07-09:
It might matter, but the trade-off has to be made somewhere, so it's a reasonable trade-off. Thanks.

closed 2019-07-09 by zostay