adafruit/Adafruit_CircuitPython_seesaw #53: seesaw_crickit_test.py fails with Invalid touch pin

veryalien, 2020-09-18:
seesaw_crickit_test.py fails with a ValueError Invalid touch pin

This is not surprising as the code tries to use ss.touch_read with crickit "pins" range(4) - 0 to 3 (no pin mapping).
In the Crickit.Pinmap the touch "pins" are correctly defined as 4 to 7, so the touch_read fails with a ValueError.
How did this example ever work?

veryalien, 2020-09-19:
After further investigation it looks like real issue is inconsistent seesaw touch "pin" numbering.
The microbit seesaw driver for some reason uses touch pins 0 to 3. All other seesaw "pins" for motors, signals, servos, etc. are the same as other seesaw drivers.
If you sonehow tested the seesaw_crickit_test with a development seesaw driver then it might explain the incorrect 0 to 3 numbering instead of 4 to 7.
However the example still doesn't work out of the box.

ladyada, 2020-09-19:
what hardware are you using, and what example are you using?

veryalien, 2020-09-19:
I'm tinkering with perhaps strange combinations. I'm using the following:
Crickit featherwing for any feather - although I do not actually have a feather!
Trinket M0 - I thought it would not matter as everything is over I2C - SDA, SCL, 5V from Crickit Board to BAT, and GND.
The I2C connection works fine, it's not exactly rocket science!

There's not too much memory in the trinket and it doesn't like the adafruit_crickit class:

Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit Trinket M0 with samd21e18
>>> 
>>> 
>>> import adafruit_crickit
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_crickit.py", line 58, in <module>
  File "adafruit_motor/stepper.py", line 60, in <module>
MemoryError: memory allocation failed, allocating 136 bytes

So, I found the Adafruit_CircuitPython_seesaw github repository and the seesaw_crickit_test example.

The library loads on the trinket and I can communicate with the crickit just fine over I2C. I took some of the example code to read the touch pins just as a test. It didn't work, it generated the Invalid touch pin error. I modified it to use touch pins 4 to 7 and can now read those pins and light corresponding leds connected to the signal pins on the crickit to 'see' a touch.

However I've now tried using some motors in a similar way to the example, but it's not working because analog_writes are apparently only 1 byte so values over 255 don't work, at least that's what the code is returning as an error. I now suspect that the seesaw_crickit_test.py is not what it seems and I've perhaps misunderstood what it is for?

veryalien, 2020-09-19:
Just to prove (to myself) that I'm not going completely crazy and I kind of know what I'm doing,  I've swapped out the Trinket M0 with an ItsyBitsy M4 Express where the adafruit_crickit library loads.

And.... everything on the crickit board now works completely as expected!

I'm really not sure what the adafruit_circuitpython_seesaw library is all about as the basic I2C comms and simple I/O signals work fine. But all the PWM stuff seems to be 'broken'.

I can obviously live with using the ItsyBitsy, but I thought I could put a currently unused Trinket M0 onto a crickit board and get a fully-functional robot up and running in no time.
.

ladyada, 2020-09-19:
well the trinket m0 will def not have enough RAM for the crikit library. the itsy m4 is best to use!

veryalien, 2020-09-19:
I will close this issue, even though the original touch pin problem is not entirely solved.
It's not worth solving a problem that doesn't really exist when the relevant boards (feather, microbit, circuit playground or raspi) are used with their respective crickits.

veryalien, 2020-09-20:
Just to make sure I wasn't dreaming, I went back to the trinket M0 connected to the featherwing crickit.

I erased the trinket and re-installed circuitpython  5.3.1 and the correct adafruit_seesaw, adafruit_motor and adafruit_bus_device libraries. I've ignored the top-level crickit convenience class as it is too large for the trinket memory.

If I do everything long-hand to create the correct PWMOut objects for drives, servos and motors it all works fine. 
Also the signal I/O and touch pins are all working. Neopixels and sound are nice-to-haves and I haven't tested them.

Of course the seesaw touch pins are 4 to 7, as you would expect with the seesaw crickit pinmap!

veryalien, 2020-09-20:
Replace line 72 of seesaw_crickit_test.py:
val = ss.touch_read(i)

With:
val = ss.touch_read(i+4)

And it will all 'magically' start working!