adafruit/Adafruit_CircuitPython_seesaw #55: Help request: Understanding Temperature Reading

psteininger, 2020-11-19:
Hi all,

I am porting parts of the Seesaw Python code to an Elixir library specifically for the soil moisture sensor. The moisture reading is straight forward, however I see something I don't fully understand in the code. Specifically Line 362 below.
Why are we doing binary AND on the first byte and why is the value `0x3f` ? 
https://github.com/adafruit/Adafruit_CircuitPython_seesaw/blob/9d91ec849efb5fbc3d26d350a75a61f15d19bc48/adafruit_seesaw/seesaw.py#L362-L363

When pulling data via I2C read in Elixir I see only "0" in the first byte, so the `&` does nothing. Also, why is the value unpacked with ">I" not ">H" as the moisture reading?

I would really like to understand these to properly port this to Elixir for the Nerves Platform

Thanks :)

ladyada, 2020-11-19:
hi its been a while but here is the C code version
https://github.com/adafruit/Adafruit_Seesaw/blob/master/Adafruit_seesaw.cpp#L659
and here is the code running on the seesaw
https://github.com/adafruit/seesaw/blob/bdc84a69b9c452d90d4eb3644d2337a5f356469b/source/Delegate.cpp#L200

psteininger, 2020-11-20:
Hi @ladyada , thank you for sending the pointers. It's been a long time since I worked with C and C++ (I think it was freshman year in college), but I think I should manage to figure things out. 

Elixir has a very interesting way of working with bits (i.e. a string is considered a "binary", because it's divisible by 8), so some things are easier than others.
For example, calling `write_read/4` returns a `BitString`, which is a binary:
```elixir
iex(211)> {:ok, raw_temp } = Circuits.I2C.write_read(ref, 54, <<0x00,0x04>>, 4)
{:ok, <<0, 24, 204, 203>>}
```
and then I can pattern match to split the parts and re-assemble if need be:
```elixir
iex(212)> <<a, b, c, d>> = raw_temp                                            
<<0, 24, 204, 203>>
iex(213)> temp1 = <<a &&& 0x3f, b, c, d>>                                      
<<0, 24, 204, 203>>
```
and then turn that `BitString` into an unsigned integer:
```elixir
iex(214)> <<temp::unsigned-integer-size(32)>> = temp1                          
<<0, 24, 204, 203>>
iex(215)> 0.00001525878 * temp                                                 
24.79995780498
```

In any case, thanks for the references, and I'll post a link here to the Elixir lib, when I get something working :)

ladyada, 2020-11-20:
awsome :)