adafruit/Adafruit_CircuitPython_seesaw #63: Should the product id be verified?

jerryneedell, 2021-05-12:
Just a thought...
Should the seesaw library also test the seesaw product id to make sure it is compatible with the device -- the Arduino code does this. For example the encoder build should be 4991
https://github.com/adafruit/seesaw/blob/master/boards/encoder/board_config.h#L4
It is checked in the Arduino example, not in the library https://github.com/adafruit/Adafruit_Seesaw/blob/master/examples/encoder/encoder_basic/encoder_basic.ino#L36

ladyada, 2021-05-12:
ill note i havent really verified older seesaw items have this PID burned in - i am verifying going forward tho!

jepler, 2021-05-13:
I added printing the pid to the simpletest, does that scratch the itch @jerryneedell ?

jerryneedell, 2021-05-13:
Yes! Thank you. Would you like to close this or keep it open for future defy?

closed 2021-05-13 by jepler