adafruit/Adafruit_CircuitPython_seesaw #77: Setting PWMOut.duty_cycle

pshanesmith, 2021-08-09:
This comments indicate a 16-bit number is expected.  If you attempt to set it to a number higher than 255, it throws an error on analog_write.  The documentation states these are 8-bit PWMs, so I believe the comments and error checking are incorrect.

`
Traceback (most recent call last):
  File "seesaw_readadc.py", line 48, in <module>
    pwm1.duty_cycle = 0xFFFF
  File "/home/pi/Testing/venv/lib/python3.7/site-packages/adafruit_seesaw/pwmout.py", line 47, in duty_cycle
    self._seesaw.analog_write(self._pin, value)
  File "/home/pi/Testing/venv/lib/python3.7/site-packages/adafruit_seesaw/seesaw.py", line 348, in analog_write
    cmd = bytearray([self.pin_mapping.pwm_pins.index(pin), value])
ValueError: byte must be in range(0, 256)
`

caternuson, 2021-08-09:
`This comments indicate a 16-bit number is expected.`

Can you link to where this is located?

pshanesmith, 2021-08-29:
Hi Cater.  Line 37 and 46 in PWMOut.py reference this.

Neradoc, 2022-05-01:
The documentation indicates:
https://github.com/adafruit/Adafruit_CircuitPython_seesaw/blob/b4cb661e6997782b68d2fb027993f9f1dc47825a/adafruit_seesaw/pwmout.py#L36-L40

The code also does:
https://github.com/adafruit/Adafruit_CircuitPython_seesaw/blob/b4cb661e6997782b68d2fb027993f9f1dc47825a/adafruit_seesaw/pwmout.py#L43-L46

But it actually depends of the chip:
https://github.com/adafruit/Adafruit_CircuitPython_seesaw/blob/b4cb661e6997782b68d2fb027993f9f1dc47825a/adafruit_seesaw/seesaw.py#L348-L351

It would be good to either unify the duty_cycle value to 0-65535, or fix the setter test and documentation to match.
```py
pwm_pin = PWMOut(ss, 4)
pwm_pin.duty_cycle = 0x8000
```
```py
Traceback (most recent call last):
  File "code.py", line 38, in <module>
  File "adafruit_seesaw/pwmout.py", line 47, in duty_cycle
  File "adafruit_seesaw/seesaw.py", line 351, in analog_write
OverflowError: value must fit in 1 byte(s)
```