adafruit/Adafruit_CircuitPython_seesaw #78: Sampling Rate for ADC is extremely low

pshanesmith, 2021-08-09:
As currently written, it appears you can only do about 93 SPS.  It looks like if there is a shorter delay introduced on analog_read, it drastically increases the sampling rate and it seems to be running stable.

`
    def analog_read(self, pin):
        """Read the value of an analog pin by number"""
        buf = bytearray(2)
        if pin not in self.pin_mapping.analog_pins:
            raise ValueError("Invalid ADC pin")

        self.read(
            _ADC_BASE,
            _ADC_CHANNEL_OFFSET + self.pin_mapping.analog_pins.index(pin),
            buf,
            .0001,
        )
        ret = struct.unpack(">H", buf)[0]
        time.sleep(0.001)
        return ret
`

caternuson, 2021-08-09:
That's probably there to enforce the underlying requirement of the seesaw firmware:
https://learn.adafruit.com/adafruit-seesaw-atsamd09-breakout/analog-to-digital-converter
```
Allow a delay of at least 1ms in between sequential ADC reads on different channels.
```

pshanesmith, 2021-08-29:
Thanks Cater.  In this case, the 1ms delay is being used between reads from the ADC on the same channel with no reads on other channels.  I was able to update that number to a much shorter delay and it could run stable.  I suppose there would need to be some detection for a channel swap I order to honor thale required delay while supporting faster reading on a single channel.

todbot, 2021-12-10:
To address this somewhat, in PR #90 I allow you to specify the delay in `seesaw.analog_read()` that's then passed along to `seesaw.read()`. So if you know your application can support faster reads, you can tune down the delay at least.