adafruit/Adafruit_CircuitPython_seesaw #80: Hi, Can we rename seesaw library element digitalio to digitalio2?

snarflakes, 2021-08-13:
It is conflicting with my adafruit screen's digitalio.  The seesaw "digitalio" i guess doesn't have all the same elements needed for launching the adafruit 1.3 TFT Bonnet screen.

Traceback (most recent call last):
  File "gifwebrotary.py", line 44, in <module>
    cs_pin = digitalio.DigitalInOut(board.CE0)
AttributeError: module 'adafruit_seesaw.digitalio' has no attribute 'DigitalInOut'

FoamyGuy, 2021-08-13:
Can you the rest of your `gifwebrotary.py` code file, or a link to it if it's already hosted in github or somewhere else online? 

It should be possible to modify your imports slightly to get it so your code will use the correct implementation of digitalio even if the seesaw stuff is also needed for other things. I can try to help you get it fixed.

snarflakes, 2021-08-15:
Thanks so much, you pointed me in the right direction.  I moved things around and got it to work. I will resubmit if I run into further problems with this conflict.

closed 2021-08-15 by snarflakes