adafruit/Adafruit_CircuitPython_seesaw #92: Can't use addr 0x38 for STEMMA Soil Sensor on Funhouse

GH15ADF, 2022-01-28:
I have a Funhouse and a few Adafruit STEMMA Soil Sensor - I2C Capacitive Moisture Sensors. The Funhouse updated to CP 7.1.1 and circup-ed all the libraries. But the adafruit_seesaw.seesaw library is balking at 0x38 as an address for the sensor. As per the Learn guide, the address can bet set to 0x36 thru 0x39. When the address is 0x36, 0x37, or 0x39 everything works fine and I get readings. This is confirmed for all three of the sensor boards I have. So it seems unlikely it is a manufacturing or test defect. Even more surprising is that I get this error even if there is **NO** board connected to the STEMMA connector when that address value is 0x38. I went back to the basics for I2C and that sensor (from the Learn guides) and this simple test program:

```
from adafruit_seesaw.seesaw import Seesaw
import board
import time

i2c_bus = board.I2C()

while not i2c_bus.try_lock():
    pass

try:
  print(
      "I2C addresses found:",
      [hex(device_address) for device_address in i2c_bus.scan()],
  )
  time.sleep(2)

finally:  # unlock the i2c bus when ctrl-c'ing out of the loop
    i2c_bus.unlock()

ss = Seesaw(i2c_bus, addr=0x38)

while True:
    # read moisture level through capacitive touch pad
    touch = ss.moisture_read()

    # read temperature from the temperature sensor
    temp = ss.get_temp()

    print("temp: " + str(temp) + "  moisture: " + str(touch))
    time.sleep(1)
```
When the address is 0x38, I get:
```
code.py output:
I2C addresses found: ['0x38', '0x77']
Traceback (most recent call last):
  File "code.py", line 20, in <module>
  File "/lib/adafruit_seesaw/seesaw.py", line 151, in __init__
RuntimeError: Seesaw hardware ID returned (0x18) is not correct! Expected 0x55 or 0x87. Please check your wiring.
```

ladyada, 2022-01-29:
yah the AHT20 is 0x38 - you can't have two items with the same address - this is normal! :)
https://www.adafruit.com/product/4566
https://learn.adafruit.com/i2c-addresses

closed 2022-01-29 by ladyada

GH15ADF, 2022-01-29:
Understood. However, the error returned is so very unhelpful, "Please check your wiring"