adafruit/Adafruit_CircuitPython_seesaw #93: two encoders at once?

troylball, 2022-02-01:


ladyada, 2022-02-01:
@jedgarpark or @kattni may already have an example they can add

jedgarpark, 2022-02-01:
I built a version of this in Arduino, but I can work on a CircuitPython example.

troylball, 2022-02-01:
So what are my next steps?  Search their git hub repositories and reach out to them directly?  Can I / should I refer to this email and say that Adafruit suggests I reach out?

jedgarpark, 2022-02-01:
@troylball I'll try to put together an example later today.

troylball, 2022-02-01:
Thanks John, that would be appreciated,  it does not need to bee too intrecit.  My end goal is to use these with a TEA 5767 FM chip to convert an old radio.  One button volume one button frequency.  I am guessing I could run two programs at the same time but thought it would nice to see it all together.

I see your other e-mail came in thanks for that

Thanks,

jedgarpark, 2022-02-02:
@troylball here's an example of two encoders https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/Seesaw_QT_Rotary_Encoder_Multiples/code.py

ladyada, 2022-02-02:
@jedgarpark  this should go into this library as an example, @kattni can help if needed

closed 2022-02-02 by ladyada

jedgarpark, 2022-02-02:
it is now in the library examples