adafruit/Adafruit_CircuitPython_seesaw #97: Make AnalogIn.value return always 16 bit?

caternuson, 2022-03-12:
The CP standard for `AnalogIn.value` is 16 bit:
https://docs.circuitpython.org/en/latest/shared-bindings/analogio/index.html#analogio.AnalogIn.value

This library wants to follow same convention:
https://github.com/adafruit/Adafruit_CircuitPython_seesaw/blob/fd26309627c0758c982d272b6c708322102bddbf/adafruit_seesaw/analoginput.py#L34

[NeoSlider](https://www.adafruit.com/product/5295) (others?) return 10 bit:
```python
Adafruit CircuitPython 7.1.1 on 2022-01-14; Adafruit ItsyBitsy M4 Express with samd51g19
>>> import board
>>> from adafruit_seesaw.seesaw import Seesaw
>>> from adafruit_seesaw.analoginput import AnalogInput
>>> neoslider = Seesaw(board.I2C(), 0x30)
>>> potentiometer = AnalogInput(neoslider, 18)
>>> potentiometer.value
1023
>>> potentiometer.value
943
>>> potentiometer.value
451
>>> potentiometer.value
0
>>> 
```

ladyada, 2022-03-13:
it shooould yes! dox need updating too

caternuson, 2022-03-14:
OK, pretty easy fix, just a matter of scaling.

Which dox?

dhalbert, 2024-03-25:
making it scale to 65535 as opposed to just 16 bit is a little trickier:
https://github.com/adafruit/circuitpython/pull/6426. See the computational trick there to fill in the low-order bits properly.