adafruit/Adafruit_CircuitPython_seesaw #105: seesaw.neopixel is incompatible with regular neopixel protocol

MauiJerry, 2022-07-08:
There is a fundamental incompatibility between regular Neopixel(and Dotstar) protocol and the seesaw version.  The common protocol is derived from PixelBuf which keeps an array of values and transmits on show(). The seesaw version does not keep the pixel values local.  It transmits them individually on each __setitem__().  This has two issues:
  * first is  it eliminates reuse of usual animation code (led_animation or other) on a seesaw,
  * second it forces individual writes to the i2c bus and seesaw device.  A fill will send individual commands for each pixel in the strip.   Most seesaw boards only have a small number of pixels (16-64) but some have up to 1024.  However, even an all pixel fill will take 128ms (at 8ms each), or over 1/10 second, which is the accepted response time for interactions.

Some Possible Solutions:
 * A block write of N pixels might alleviate the responsive issue.  I2C itself does not limit the message size, but code libraries often do put a limit on them.  (16pix * 3byte/pix + header is still < 64 bytes). Need to check against adafuit.seesaw code for supported boards.
 * A layer over seesaw.neopixel that kept a 'host' side array (derived from PixelBuf) could help make animations work with seesaw devices.
  * create whole new seesaw.neopixel animation library

ladyada, 2022-07-10:
we do want to tweak neopixel support but may not get to it ASAP - any optimizaiton code to this library is appreciated as a PR

MauiJerry, 2022-07-11:
Neradoc shared a prototype PixelBuf derived seesaw.neopixel replacement.  Unsure of how tested it is. (attached with some unfinished comments by me)
[neopixel-subclass.py.txt](https://github.com/adafruit/Adafruit_CircuitPython_seesaw/files/9087965/neopixel-subclass.py.txt)

dhalbert, 2022-07-22:
@MauiJerry would you like to test #106, or have you already done so? If it's fine, we'll make a major release. Thanks!

closed 2022-07-22 by dhalbert

MauiJerry, 2022-07-23:
i used Neradoc's branch in our project on rPi Pico feather with no
problems. not an exhaustive test but it ran at an event for 6-7hr.

- jerry isdale

sent from my phone, which may have added its own interpretation of what i
typed or dictated. I apologize for any auto correct errors introduced by
our robotic under lord.