adafruit/Adafruit_CircuitPython_seesaw #113: Seesaw Capacitive Moisture Sensor's highly random data output, maxing out at 1015 rather then 2000 and maxing out super soon even in dry soil.

Sandesh4242, 2023-05-06:
Hello Everyone, 

I have been using seesaw soil moisture for some period of time now, everything works out fine, the code, the library everything. I developed two identical systems based on raspberry pi with each soil moisture sensor along with a couple more sensors to be placed in two identical greenhouses near my house. 

When I was developing the system, everything looked fine, the data was also promising. Whenever the soil moisture sensor was in the air the data would be around 300-400, when I would touch it it would go above 1000. 

Once I placed the sensor in the field, I quickly came to realize that the data was not precise and highly random. 

The issues,
1.  The data is not accurate, even if you place a high delay between sensor reading and the average of 10 or more data.
2. The data never goes above 1015, tested this in raspberry pi 4 as well as esp32-devkit. (if the data is supposed to max out at 
    1015, why does Adafruit's website claim to get 2000, maybe they should change that as well on the official website.)
3. I would still be happy if the data would change from 300-1015 in a progressive way based on soil moisture content but it 
   doesn't, I have placed these sensors in really moist soil and really dry soil, and the readings are almost similar.  The conditions 
   for me are not some soil placed in a pot, I am talking about a natural field. when I place the sensor in the soil, the reading is 
   always maxed out, no matter whether it is dry soil or moist.  if I remove it data goes back to less than 400, if I plug the sensor 
  in the soil, the data is above 1000. 

I am an electrical engineer myself, been using Adafruit's hardware for six years, so far I have never come across something bad like this from the Adafruit team. I have checked the hardware design as well, the design looks good and solid, however, if the sensor is just built to demonstrate data change between air and touch then maybe it's not a moisture sensor after all. 

I would be honored if someone could address this issue. The sensor is not useable in real cases at the moment.

ulrikkau, 2024-04-07:
I am realizing the same problem, and in water it goes down to 600. Was there any progress?

Sandesh4242, 2024-04-26:
Unfortunately, despite my attempts to reach out to the Adafruit team and others regarding this issue, I regrettably did not receive any response. Consequently, I decided to transition to an alternative type of resistive soil moisture sensor.

kaiichang, 2026-10-17:
I experience the same issue. I had to compensate by giving low values higher weight. Which sensor did you switch to? @Sandesh4242