adafruit/Adafruit_CircuitPython_seesaw #121: MIssing explanation of why Pin 24 is required for rotary encoders

erichiggins, 2023-10-01:
All of the example code and learning pages for the rotary encoder include these lines:

```
seesaw.pin_mode(24, seesaw.INPUT_PULLUP)
switch = digitalio.DigitalIO(seesaw, 24)
```
but I cannot find any documentation that explains why it's required. The examples don't work without it (I've tried) and nothing is connected to pin 24 (GPIO24?) in the wiring diagrams. I'd love to understand what is going on here and why this is needed.

I'm trying to get this rotary encoder working on a RP2040 with Micropython. My guess is this mystery may be what is blocking me.

Thanks!

erichiggins, 2023-10-02:
Products I'm using:

1. Raspberry Pi Pico
2. [Rotary Encoder + Extras](https://www.adafruit.com/product/377)
3. [Adafruit I2C Stemma QT Rotary Encoder Breakout with NeoPixel - STEMMA QT / Qwiic](https://www.adafruit.com/product/4991)
4. [Adafruit 2.23" Monochrome OLED Bonnet for Raspberry Pi](https://www.adafruit.com/product/4567)  
    Connected via jumper wires to the Raspberry Pi Pico (I2C0)

When I have the rotary encoder and breakout board connected to the OLED display via STEMMA cable, I do detect it when I run `i2c.scan()` and can confirm it is not detected when the cable is disconnected.

> In general, those commands are setting pin mode on the seesaw chip itself. The reasoning would depend on the hardware specifics of what encoder is used and how it is connected to the seesaw chip.

That's interesting. On the Learning page for this product, Pin 24 is used for a Raspberry Pi, a Feather board, and an Arduino. Is it coincidence that they all happened to be Pin 24 on those products?
https://learn.adafruit.com/adafruit-i2c-qt-rotary-encoder?view=all

caternuson, 2023-10-02:
The Pin 24 being reference is on the seesaw chip itself, which is the SAMD09 microcontroller on the I2C Stemma QT breakout. See the schematic in that Learn guide. It's the pin here:
[image]
which is connected to the push button switch on the rotary encoder. That switch gets closed when the encoder knob is pressed down. The fact that it's pin 24 is a detail buried in the seesaw firmware. But regardless of main board being used (Pi, feather, etc.) and regardless of programming (CircuitPython, Arduion, etc.), it would still be reference as pin 24.

The reason for enabling the internal pull up resistor has to do with dealing with floating inputs.

erichiggins, 2023-10-02:
Ok, that makes sense. It sounds like I may need to find an equivalent way to do this since I'm not using the seesaw library.

From your perspective, does it make sense to include this detail in the documentation somewhere?

caternuson, 2023-10-02:
Yah, it is sort of a buried detail. I think the general idea is the code examples provide what is needed.

erichiggins, 2023-10-02:
I get it. Even just adding a simple inline code comment about what it's doing would help to clarify.

caternuson, 2023-11-07:
@erichiggins Made a PR to add a code comment. Does that wording help?
https://github.com/adafruit/Adafruit_CircuitPython_seesaw/pull/124/files
(can add comments in PR thread if you have better suggestions)

erichiggins, 2023-11-07:
@caternuson That helps, thanks!

caternuson, 2023-11-07:
No problem. Comment added and merged.

closed 2023-11-07 by caternuson