adafruit/Adafruit_CircuitPython_seesaw #127: FT232H Connected to Neopixel Drive gives an Invalid hardware  ID

scirelli, 2024-01-30:
I have a [FT232h](https://learn.adafruit.com/adafruit-ft232h-breakout) connected to a [Neopixel Driver board](https://learn.adafruit.com/adafruit-neodriver-i2c-to-neopixel-driver/overview). They are connected  via the FT232H STEMMA to the NeoPixel Drivers STEMMA connector.

[Example code](https://learn.adafruit.com/adafruit-neodriver-i2c-to-neopixel-driver?view=all#simple-test-example-code-3146428) I'm testing with.

Error being raised:
```
Traceback (most recent call last):

  File "./i2c_neopixel.py", line 32, in <module>

    ss = seesaw.Seesaw(i2c, addr=0x60)

         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "./.venv/lib/python3.11/site-packages/adafruit_seesaw/seesaw.py", line 161, in __init__

    raise RuntimeError(

RuntimeError: Seesaw hardware ID returned 0x67 is not correct! Please check your wiring.
```

The guide has me installing [adafruit-circuitpython-seesaw](https://github.com/adafruit/Adafruit_CircuitPython_seesaw/tree/main) I see where the error is being raise on lines [152-164](https://github.com/adafruit/Adafruit_CircuitPython_seesaw/blob/main/adafruit_seesaw/seesaw.py#L152-L164).
`pip install adafruit-circuitpython-seesaw`

Is a new chip being  used on the NeoPixel driver board, and the code  hasn't been updated?

There are some other [examples](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/i2c#run-example-3041249) that setup the I2C with 
```
i2c = board.STEMMA_I2C()
```
I modified the [example](https://learn.adafruit.com/adafruit-neodriver-i2c-to-neopixel-driver?view=all#simple-test-example-code-3146428) code I was using to test the FT232H and NeoPixel driver to use the above but that gives me this error:
```
Traceback (most recent call last):
  File "./i2c_neopixel.py", line 32, in <module>
    i2c = board.STEMMA_I2C()  # For using the built-in STEMMA QT connector on a microcontroller
          ^^^^^^^^^^^^^^^^
AttributeError: module 'board' has no attribute 'STEMMA_I2C'
```

scirelli, 2024-01-30:
@ladyada where do you get the [chip ids](https://github.com/adafruit/Adafruit_CircuitPython_seesaw/blob/main/adafruit_seesaw/seesaw.py#L107-L113) from? I'd like to update the list, if `0x43` is a valid chip id. Thanks!

caternuson, 2024-01-30:
It's baked into the firmware.

For ATtiny based seesaws (neodriver is one of these), the firmware is here:
https://github.com/adafruit/Adafruit_seesawPeripheral
and `HW_ID` is set via preproc logic:
https://github.com/adafruit/Adafruit_seesawPeripheral/blob/313090f3a3839e1e5f8f0d7a81c6a6262f89f571/Adafruit_seesawPeripheral_request.h#L1-L15

For SAMD based seesaws, the firmware is here:
https://github.com/adafruit/seesaw

The value itself comes from querying the Status function register for `HW_ID`:
https://learn.adafruit.com/adafruit-seesaw-atsamd09-breakout/status

`0x43` is not a valid chip id. Something else is happening and will need to be investigated. It may also be something specific to the FT232H.

scirelli, 2024-01-30:
Is this something I can help with? I'm now sure what I can do. 
I have  the FT232H hooked up to a Mac.

caternuson, 2024-01-30:
>  I had to reduce the frequency for it to work though.

That's actually probably the best fix for getting it working with the FT232H. There is potentially a small amount of clock stretching happening with the seesaw chip, and the FT232H may or may not be handling that well. Slowing down the I2C frequency is a good work around.

With the slowed down clock, the id is probably reading correctly now as well. Can verify via REPL:
```python
>>> import board, busio
>>> from adafruit_seesaw import seesaw, neopixel
>>> i2c = busio.I2C(board.SCL, board.SDA, frequency=10000)
>>> ss = seesaw.Seesaw(i2c, addr=0x60)
>>> ss.chip_id
135
>>> hex(_)
'0x87'
>>>
```

scirelli, 2024-01-31:
> That's actually probably the best fix for getting it working with the FT232H. There is potentially a small amount of clock stretching happening with the seesaw chip, and the FT232H may or may not be [...]

Yes, now it does read the correct chip id! I'll close that other PR with a note.

scirelli, 2024-01-31:
Can we add a note to the example code that the frequency may need to be slowed down? I can do it if you point me to where it lives.