adafruit/Adafruit_CircuitPython_seesaw #128: Error message prints wrong number system

scirelli, 2024-01-30:
[Lines 161-164](https://github.com/adafruit/Adafruit_CircuitPython_seesaw/blob/main/adafruit_seesaw/seesaw.py#L161-L164)
In the raised error the message adds a `0x` to the `self.chip_id` which is in decimal. In my case the error message read

```
RuntimeError: Seesaw hardware ID returned 0x67 is not correct! Please check your wiring.
```
The `67` decimal, which is `0x43` in hex.

caternuson, 2024-01-30:
Looks like this happened when changing to F string format done in this PR:
https://github.com/adafruit/Adafruit_CircuitPython_seesaw/pull/118/files

The previous `format` syntax included `{:x}` to format the result as hex. A similar format specifier is missing in the current library's f-string syntax.

It's a simple fix though:
```python
>>> foo = 67
>>> f"0x{foo}"
'0x67'
>>> f"0x{foo:x}"
'0x43'
>>> 
```

scirelli, 2024-01-30:
I can create a PR to fix it