adafruit/Adafruit_CircuitPython_seesaw #134: call to seesaw.get_options() causes Remote I/O Error (Errno 121)

PaulskPt, 2024-02-22:
I wrote a python script to use an Adafruit Gamepad QT (Prod ID 5743) to a Raspberry Pi 5B-8GB.
Installed Blinka and all other requirements. Everything is running fine except that a call to function seesaw.get_options causes a ```Remote I/O Error (Errno 121)```. To see what is happening I added a print statement inside the function seesaw.read().

Print output of my Python script:

```

(env) paulsk@raspi5B-8GB:~/Adafruit_GamepadQT $ python3 gamepadqt.py
Board id:      RASPBERRY_PI_5
Board serial:  1169efc3bd80e900
Board rev:     d04170
env/lib/python3.11/site-packages/adafruit_seesaw/seesaw.py/seesaw.read()
        params reg_base: 0x0, reg: 0x1, buf: bytearray(b'\x00')                           <<<=== _STATUS_HW_ID
env/lib/python3.11/site-packages/adafruit_seesaw/seesaw.py/seesaw.read()
        params reg_base: 0x0, reg: 0x2, buf: bytearray(b'\x00\x00\x00\x00')               <<<=== _STATUS_VERSION 
env/lib/python3.11/site-packages/adafruit_seesaw/seesaw.py/seesaw.read()
        params reg_base: 0x0, reg: 0x2, buf: bytearray(b'\x00\x00\x00\x00')               <<<=== _STATUS_VERSION 
env/lib/python3.11/site-packages/adafruit_seesaw/seesaw.py/seesaw.read()
        params reg_base: 0x0, reg: 0x3, buf: bytearray(b'\x00\x00\x00\x00')               <<<=== _STATUS_OPTIONS
seesaw.read(): Error: [Errno 121] Remote I/O error                                        <<<=== The error
gamepad_test(): seesaw.chip_id= 0x87
gamepad_test(): seesaw._version()= 0x166f7a97 hex
gamepad_test(): seesaw.get_options()= 0x0 hex
gamepad_test(): We're going to test the gamepad.
                Press a button on the gamepad...
                To reboot the Raspberry Pi press gamepad button Start
```

To prevent the script crashes because of the error I added a try...except block inside the function seesaw.options(), as follows:
```
def get_options(self):
        """Retrieve the 'options' word from the SeeSaw board"""
        buf = bytearray(4)
        try:
            self.read(_STATUS_BASE, _STATUS_OPTIONS, buf)
        except OSError as e:
            if e.errno == 121:  # remote I/O Error
                print(f"seesaw.get_options(): Error: {e}")
                pass
        ret = struct.unpack(">I", buf)[0]
        return ret
```

QUESTION:
I would like to know what is causing this error.

caternuson, 2024-02-22:
Native CP also has issues:
```python
Adafruit CircuitPython 8.2.10 on 2024-02-14; Adafruit QT2040 Trinkey with rp2040
>>> import board
>>> from adafruit_seesaw.seesaw import Seesaw
>>> i2c = board.STEMMA_I2C()
>>> seesaw = Seesaw(i2c, addr=0x50)
>>> seesaw.get_version()
376404631
>>> seesaw.analog_read(14)
506
>>> seesaw.get_options()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_seesaw/seesaw.py", line 205, in get_options
  File "adafruit_seesaw/seesaw.py", line 508, in read
OSError: [Errno 19] No such device
>>> 
```

caternuson, 2024-02-22:
Nothing weird in I2C traffic (using same QT2040 Trinkey from above).

Here's the call to `seesaw.get_options()`:
[image]
That's the initial write request, a ~8ms sleep [which is coming from the library](https://github.com/adafruit/Adafruit_CircuitPython_seesaw/blob/3109848de6c7de1755ac2ccdcfc58460b7cc09bf/adafruit_seesaw/seesaw.py#L506), and then the attempt to read the 4 bytes.

The write is what is should be:
[image]
but the read attempt is NAK'd right away:
[image]

That would seem to indicate something more with the firmware itself.

caternuson, 2024-02-22:
This is a limitation of the current version of the seesaw firmware being used on the Adafruit Gamepad QT (Prod ID 5743).

The `OPTIONS` function of the `STATUS` base register has not been implemented yet:
https://github.com/adafruit/Adafruit_seesawPeripheral/blob/313090f3a3839e1e5f8f0d7a81c6a6262f89f571/Adafruit_seesawPeripheral_request.h#L25-L31
Only `HW_ID` and `VERSION` have been implemented.

Note that an attempt to read the `TEMP` register also fails in a similar way:
```python
>>> seesaw.get_temp()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_seesaw/seesaw.py", line 386, in get_temp
  File "adafruit_seesaw/seesaw.py", line 508, in read
OSError: [Errno 19] No such device
>>> 
```

@PaulskPt What is your use case for needing to query the OPTIONS register?